grblHAL/ESP32 #109: Build fails with error in uart_serial.c

melvinisken, 2024-04-23:
I just cloned the project and tried to compile it using ESP IDF 4.3.1 (is that still the correct version?). I get the following error (the same if I try via Visual Studio Code or IDF command line:

```
../main/uart_serial.c: In function '_uart_ll_get_txfifo_count':
../main/uart_serial.c:393:12: error: implicit declaration of function 'HAL_FORCE_READ_U32_REG_FIELD' [-Werror=implicit-function-declaration]
     return HAL_FORCE_READ_U32_REG_FIELD(hw->status, txfifo_cnt);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main/uart_serial.c:393:53: error: 'txfifo_cnt' undeclared (first use in this function)
     return HAL_FORCE_READ_U32_REG_FIELD(hw->status, txfifo_cnt);
                                                     ^~~~~~~~~~
../main/uart_serial.c:393:53: note: each undeclared identifier is reported only once for each function it appears in
../main/uart_serial.c:394:1: error: control reaches end of non-void function [-Werror=return-type]
 }
```
Did I miss something? I changed my machine config to MKS DLC32 board, but no other relevant changes, I think.

terjeio, 2024-04-24:
I am able to build with 4.3.6 - or am I?
I am using Espressif-IDE and idf.py to build and have several frameworks installed, the IDE allows switching between them. Looking closer it seems switching may not actually work as it should - a build might be mixing framework sources. So a clean 4.3.x install may not work? Anyway I am using 4.4.6 now.

melvinisken, 2024-04-24:
I just installed a new version, 4.4.7 (latest stable 4, I guess). Now the serial part does compile, but I get another error. I had no time to dig deeper, I will have a closer look later.

```
[...]
[98/100] Linking C executable bootloader.elf
[99/100] Generating binary image from built executable
esptool.py v3.3.4-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated C:/Development/ESP32/build/bootloader/bootloader.bin
[100/100] cmd.exe /C "cd /D C:\Development\ESP32\build\bootloader\esp-idf\esptool_py && C:\Development\.espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/Development/esp-idf-v4.4.7/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Development/ESP32/build/bootloader/bootloader.bin"
Bootloader binary size 0x4350 bytes. 0x2cb0 bytes (40%) free.
[1462/1464] Linking CXX executable grbl.elf
FAILED: grbl.elf
cmd.exe /C "cd . && C:\Development\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\grbl.elf.rsp -o grbl.elf  && cd ."
c:/development/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Development/esp-idf-v4.4.7/components/esp_wifi/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.18+0x0): undefined reference to `btdm_rf_bb_reg_init'
c:/development/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Development/esp-idf-v4.4.7/components/esp_wifi/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.18+0x16): undefined reference to `btdm_rf_bb_reg_init'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
I did not remove my 4.3 version, I don't know if it creates any of such conflicts.

melvinisken, 2024-04-24:
According to https://github.com/espressif/esp-idf/issues/13113#issuecomment-1933379862 this is a coexistance problem with wifi and bluetooth. The situation is:
- if I disable wifi and bluetooth, compilation is successful
- if I enable one of them, I get the error above
- if I enable both, I don't get the error, but I get a new error that the partition is too small for grbl.bin

I will investigate further.

terjeio, 2024-04-24:
I forgot to mention the sdkconfig file, replace it with sdkconfig.4.4 - does that help?

melvinisken, 2024-04-25:
I didn't find any sdkconfig.4.4, so I just renamed the old one, a new one has been generated then, now I get another error,
```
C:/Development/ESP32/main/bluetooth.c:40:10: fatal error: esp_bt.h: No such file or directory
 #include "esp_bt.h"
```
So I guess I will try to remove all instances of the espressif sdks, install 4.4.7 again and will try again. I don't think jumping from one error to the next will help and will just take your time :-)

melvinisken, 2024-04-25:
Ok, made some progress. I just renamed sdkconfig.esp32doit-devkit-v1 to sdkconfig and now I can compile. Maybe that's what you meant before. 
The only drawback I have now is that I can't enable WebUI 3, this seems to be too large (at least in combination with WiFi, gcode clustering and laser coolant), but WebUI 2 seems to work.

terjeio, 2024-04-25:
> Ok, made some progress. I just renamed sdkconfig.esp32doit-devkit-v1 to sdkconfig and now I can compile. Maybe that's what you meant before.

Not really, I see now that the sdkconfig.4.4 file has not been commited. Don't know if it helps, here is a copy:
[sdkconfig.4.zip](https://github.com/grblHAL/ESP32/files/15112037/sdkconfig.4.zip)

[partitions.csv](https://github.com/grblHAL/ESP32/blob/master/partitions.csv) has to be changed (increase factory area) to accomodate larger code, I have not done this yet since it will move the settings area causing existing values to be lost.

And odd that you have to compile with both WiFi and Bluetooth since having both enabled is, AFAIK, causing issues.

melvinisken, 2024-04-25:
> Not really, I see now that the sdkconfig.4.4 file has not been commited. Don't know if it helps, here is a copy: [sdkconfig.4.zip](https://github.com/grblHAL/ESP32/files/15112037/sdkconfig.4.zip)

Ah, ok, that's the reason then. I just tried your 4.4 and it does not work (coexistence error). So I continue to use the esp32doit..-file. Could I get any problems later when using this one?
> [partitions.csv](https://github.com/grblHAL/ESP32/blob/master/partitions.csv) has to be changed (increase factory area) to accomodate larger code, I have not done this yet since it will move the settings area causing existing values to be lost.

That would be no problem since I could do a complete erease before. 
> And odd that you have to compile with both WiFi and Bluetooth since having both enabled is, AFAIK, causing issues.

Using the  esp32doit..-file it is no problem, I only enabled WiFi and it works. Using your 4.4-file the error is there again.

melvinisken, 2024-04-28:
Another small update: the esp32diot-file does compile, but it does not boot after flashing. I then used your 4.4 file and this works, but only without wifi/bt. I can enable wifi AND bt without errors, but if I include webui, it does not fit into the available space anymore. Disabling wifi/bt compiles and boots fine. Will test with an actual laser now. 

So it would be great to have a new partition scheme which gives more space for the app.

melvinisken, 2024-05-03:
Next update :-) I took your sdkconfig and only changed one thing, disabled the SW coexistence setting:
[image]

Now I can compile the stack with Wi-Fi only and even everything seems to fit into the space, I activated Wi-Fi, WebUI 3, sd card and more, and it is working. I will do further testing. I attached my sdkconfig as well.
[sdkconfig.4.4.7.zip](https://github.com/grblHAL/ESP32/files/15199627/sdkconfig.4.4.7.zip)