grblHAL/ESP32 #112: Controller randomly stops

auxp, 2024-06-16:
often The controller always pauses randomly for no reason, the spindle continues to rotate, the motion stops randomly, and the controller cannot be operated. It can only be powered off

terjeio, 2024-06-20:
You have an interference (EMI) issue?
If you do not power up the motor and the spindle and dry run some code does it still pause?

auxp, 2024-06-20:
> 您有干扰 （EMI） 问题吗？如果你不给电机和主轴通电并试运行一些代码，它还会暂停吗？

Yes, it's caused by interference. There shouldn't be any problem if the spindle doesn't start, but my spindle power supply and controller are separate. GND and PWM are controlled by inductors. Is there anything in the controller that can be optimized?

terjeio, 2024-06-20:
> Is there anything in the controller that can be optimized?

Which board? And have you tried different USB cables, possibly with ferrites?

auxp, 2024-06-20:
> > 控制器中有什么可以优化的吗？
> 
> 哪个板？您是否尝试过不同的 USB 电缆，可能带有铁氧体？

Thank you for the reminder! I think it may be caused by a USB cable. Currently, Type-C is being used which does not have a shielding layer. Previouslyusing USB-B did not have this problem, and the anti-interference ability of CH340 is not as good as RS232