grblHAL/ESP32 #113: unable to compile latest version

boboxx, 2024-06-20:
Just did a git pull and i'm getting this error trying to compile:

Compiling .pio/build/esp32/main/trinamic/tmc2209.o
main/spi.c: In function 'spi_put_byte':
main/spi.c:135:39: error: 'SPI_TRANS_MODE_OCT' undeclared (first use in this function); did you mean 'SPI_TRANS_MODE_QIO'?
         .flags = SPI_TRANS_USE_TXDATA|SPI_TRANS_MODE_OCT,
                                       ^~~~~~~~~~~~~~~~~~
                                       SPI_TRANS_MODE_QIO
main/spi.c:135:39: note: each undeclared identifier is reported only once for each function it appears in
Compiling .pio/build/esp32/main/trinamic/tmc2209hal.o

terjeio, 2024-06-20:
Remove SPI_TRANS_MODE_OCT and see what happens. I added this a few days ago so must be another incompatible Espressif change in the frameworks of theirs... (I am using IDF 4.4.3).

boboxx, 2024-06-20:
[ESP32](https://github.com/grblHAL/ESP32/tree/master)
/platformio.ini

Should be updated to include:
platform = espressif32 @ ~5.0.0

boboxx, 2024-06-20:
I just tried the web buildwer and it's also failling wit the same problem :)

Command failed: pio run -etest -cjob.ini

[notice] A new release of pip is available: 23.3.2 -> 24.0
[notice] To update, run: pip install --upgrade pip
main/spi.c: In function 'spi_put_byte':
main/spi.c:135:39: error: 'SPI_TRANS_MODE_OCT' undeclared (first use in this function); did you mean 'SPI_TRANS_MODE_QIO'?
.flags = SPI_TRANS_USE_TXDATA|SPI_TRANS_MODE_OCT,
^~~~~~~~~~~~~~~~~~
SPI_TRANS_MODE_QIO
main/spi.c:135:39: note: each undeclared identifier is reported only once for each function it appears in
*** [.pio/build/test/main/spi.o] Error 1
========================= [FAILED] Took 27.62 seconds =========================

closed 2024-06-24 by boboxx