grblHAL/ESP32 #117: Can't compile BOARD_MKS_TINYBEE_V1 .esp32 I get errors

regin92, 2024-07-08:
mne nuzhno dlya esp32 BOARD_MKS_TINY pomyanyat' piny . s ofitsial'noy kartoy to chto ustanavlivayu Web Builder ne vse piny pravil'no vystavleny pod lazer nuzhno ne mnogo izmenit' chto by dobavit' ustroystva dlya m8 i pomenyat' pin lazera . ustanovil Espressif . kloniroval proyekt otredaktiroval fayly ESP32\main\CMakeLists.txt ESP32\main\my_machine.h \ESP32\main\grbl\config.h \ESP32\main\boards\mks_tinybee_1_0_map.h poluchayu oshibki . sam ispravit' ne smogu . log prilagayu i fayly kotoryye ya pravil .
Ещё
496 / 5 000
I need for esp32 BOARD_MKS_TINY
change the pins.
with the official map, the fact is that I’m installing Web Builder, not all pins are correctly set for the laser; you need to change a little to add devices for m8 and change the laser pin.
installed Espressif. cloned the project
edited the files
ESP32\main\CMakeLists.txt
ESP32\main\my_machine.h
\ESP32\main\grbl\config.h
\ESP32\main\boards\mks_tinybee_1_0_map.h
I get errors. I can’t fix it myself.
I am attaching the log and the files that I edited.
[image]
[ESP32.zip](https://github.com/user-attachments/files/16126845/ESP32.zip)
[ESP32_MKS_TINYBEE_V1.json](https://github.com/user-attachments/files/16126895/ESP32_MKS_TINYBEE_V1.json)

terjeio, 2024-07-08:
Wrong IDF version? Espressif has a habit of making new version not backwards compatible, grblHAL is for 4.3 or 4.4. And as long as the versions used for grblHAL is working I do not want to spend time refactoring/retesting the code.

_CMakeLists.txt_ should not be modified unless you add a custom plugin, enable stuff in _my_machine.h_ instead.

regin92, 2024-07-09:
Thank you.
installed Espressif\frameworks\esp-idf-v4.4.8\
and everything compiled fine
 everything works but I don’t know where to turn on the limit switches and the house.
many settings. I'm still figuring it out. where can I find descriptions of the settings?
 I use LightBurn. the machine is working fine. but the house doesn't work
Previously, the house and limit switches worked on other grbl firmwares.
 I configure it via ioSender. and web browser index.html.gz .
I did several firmware updates and configured my_machine.h and here’s the question:
Is it possible to change anything in grbl/config.h?
i just changed #define COREXY 1 but i need end switches and house
#define DEFAULT_HOMING_FORCE_SET_ORIGIN Off
And
#define DEFAULT_LIMITS_TWO_SWITCHES_ON_AXES Off
can this be enabled here?

closed 2024-09-25 by terjeio