grblHAL/ESP32 #119: Spindle minimum stop time 1 sec. and I have a laser.

regin92, 2024-07-19:
Spindle minimum stop time 1 sec. and I have a laser.
the laser from figure to figure does not have time to turn off the laser. burns remain.
Minimum shutdown time in spindle settings is 1 sec. cannot be set to 0 sec. I can’t fix the code because I’m not good at it. what to do . I mastered everything else. compiled for my board. set up the house correctly. configured the entire web interface. Even the joystick works. But how to work with a laser is impossible. Fluidnc also works fine, but with grblhal Laser it works smoother and more correctly, but it does not have time to turn off.
I want to make a video on how to compile and then configure all the functions. and connect tft35 display from dlc32 to tinybee. but for now I need to figure out how to turn off the spindle instead of 4, put 0 and this is not yet possible.

terjeio, 2024-07-19:
> Spindle minimum stop time 1 sec. and I have a laser.

There is no setting for spindle stop time.
Have you enabled [laser mode](https://github.com/gnea/grbl/blob/master/doc/markdown/laser_mode.md) ($32=1)?

regin92, 2024-07-19:
yes I included it. I did it differently.
I’ll now compile and configure it in a new way and spend another 5 hours. I'll try to make a video.

regin92, 2024-07-20:
I'm not attentive. I didn't include $32=1
Now the laser is working normally. I ran tests and it does everything right. Everything is fine .
but there is a small problem. I'm adding additional content
 WebUI v3 Extension - Virtual Jog Joystick but it is not displayed on the panel and therefore it does not work. it displays and works on FluidNC.
I compiled yesterday's firmware with the latest amendments.
this is how it is in my picture. field is empty. and this is what he looks 
[image]
here's a link to what it looks like
[https://github.com/jeyeager65/ESP3D-WEBUI-Mods/tree/main/Extensions/JoyJog](url)

here is my video where it works on the FluidNC firmware there is a small piece in the middle
[https://www.youtube.com/watch?v=N77fYsrg8nc&t=321s](url)

terjeio, 2024-07-21:
> WebUI v3 Extension - Virtual Jog Joystick but it is not displayed on the panel and therefore it does not work. it displays and works on FluidNC.

You (or the author) will have to modify the extension for grblHAL since it reads some setting values from the controller.