grblHAL/ESP32 #120: Z轴不稳定？

auxp, 2024-09-28:
The Z-axis is unstable, and after working for a period of time, the Z-axis may sometimes lose steps and become higher or lower than before

terjeio, 2024-09-29:
Which board?

auxp, 2024-09-29:
I modified it myselfPCB

terjeio, 2024-09-29:
Ok, is it using RMT or I2S for the step pulses?
And how long does it typically take before steps are lost?

I assume you have already eliminated hardware issues such as binding, of configuration issues such as too high acceleration and speed?

auxp, 2024-09-29:
[image]
https://github.com/grblHAL/ESP32/issues/120

auxp, 2024-09-29:
Will acceleration affect?What I set up is 200.
maximum speed 2000.
Pulse 800。
Adjust the speed during work to 300-500%

terjeio, 2024-09-29:
> Ok, is it using RMT or I2S for the step pulses?

This is about how the ESP32 is set up: is the stepper drivers connected via shift register chips (I2S driven 74AHCT595) or directly to the processor pins?

> Will acceleration affect?

Acceleration, speed, stepper current, stepper type - all can affect the max rate. As can binding - are there any spots where the Z-axis does not travel freeely as it should?