grblHAL/ESP32 #121: XproV5 stepper settings

SpikZASA, 2024-09-29:
Hi all,

I'm really struggling to get the steppers(23HD86001Y-21B) setup properly.

They 'groan' and like bind if i jog more than 10mm. i copied the settings from my fluidnc config which works fine on fluidnc.

my setup its a bulkman 1000x1000 with xprov5 with 23HD86001Y-21B steppers, (2x y steppers 1x y limit switch) i used webbuilder with xprov5 board with y axis selected as ganged.

i've attached my config from fluidnc if that helps.
[config - Copy.txt](https://github.com/user-attachments/files/17179174/config.-.Copy.txt)

SpikZASA, 2024-09-29:
$0=4.0
$1=255
$2=0
$3=5
$4=7
$5=0
$6=1
$8=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$17=0
$18=0
$19=0
$20=1
$21=1
$22=3
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=1.0
$30=1000.000
$31=0.000
$32=0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$70=7
$73=1
$74=threeametis
$75=********
$100=159.60000
$101=160.10000
$102=400.00000
$110=5000.000
$111=5000.000
$112=1500.000
$120=100.000
$121=100.000
$122=50.000
ok
$130=760.000
$131=760.000
$132=160.000
$140=1950
$141=1950
$142=1950
$150=8
$151=8
$152=8
$180=100.0
$181=100.0
$182=100.0
$190=2000.0
$191=2000.0
$192=500.0
$200=22.0
$201=22.0
$202=22.0
$210=50
$211=50
$212=50
$220=22.0
$221=22.0
$222=22.0
$320=grblHAL
$322=192.168.5.1
$323=192.168.5.1
$324=255.255.255.0
$325=23
$326=80
$327=81
$337=
$339=0
$340=0.0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=4
$372=0
$374=2
$375=50
$384=0
$396=30
$397=0
$398=100
$460=1
$481=0
$484=1
$486=0
$650=1
$673=1.0

terjeio, 2024-09-29:
Can you reload the Web Builder and try a new build?

SpikZASA, 2024-09-29:
thank you i will do so, also which settings do i choose to make my huanyang vfd work via rs485 like in fluidnc config? it gives error 14 when i try try it on
is this fine?
[image]

SpikZASA, 2024-09-29:
> Can you reload the Web Builder and try a new build?

much better now just y2/a axis is not following Y and binds up. Spindle on still causes error 14.

SpikZASA, 2024-09-29:
correction feed rate at 5000 causes problems but at 4500 its okay. just spindle rs485 story left causes error 14

terjeio, 2024-09-30:
The serial port for RS485 was incorrectly set up in the map file, updated now. Note that the default baud rate in grblHAL is 19200.

SpikZASA, 2024-09-30:
morning i tried it again still giving that error 14. ive attached my .json settings if that helps

$0=4.0
$1=255
$2=0
$3=5
$4=0
$5=0
$6=1
$8=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=6
$15=0
$17=0
$18=0
$19=0
$20=1
$21=1
$22=9
$23=0
$24=100.0
$25=2000.0
$26=250
$27=2.500
$28=0.100
$29=1.0
$30=24000.000
$31=6000.000
$32=0
$37=7
$39=1
$40=1
$43=2
$44=4
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$70=7
$73=1
$74=threeametis
$75=********
$100=159.60000
$101=160.10000
$102=400.00000
$110=4500.000
$111=4500.000
$112=1500.000
$120=100.000
$121=100.000
$122=50.000
$130=760.000
$131=760.000
$132=160.000
$140=1500
$141=1500
$142=1500
$150=8
$151=8
$152=8
$180=100.0
$181=100.0
$182=100.0
$190=2000.0
$191=2000.0
$192=500.0
$200=22.0
$201=22.0
$202=22.0
$210=50
$211=50
$212=50
$220=22.0
$221=22.0
$222=22.0
$320=grblHAL
$322=192.168.5.1
$323=192.168.5.1
$324=255.255.255.0
$325=23
$326=80
$327=81
$337=
$339=0
$340=1.0
$341=3
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=4
$374=3
$375=250
$384=0
$396=30
$397=1000
$398=100
$460=1
$481=0
$484=1
$486=4
$650=1
$673=1.0
$H
M3 S6000
ALARM:14
#SOFTRESET#
GrblHAL 1.1f ['$' or '$HELP' for help]
$X
Caution: Unlocked]
$X
$H
M3 S6000
ALARM:14
$X
$H
[export (1).json](https://github.com/user-attachments/files/17184129/export.1.json)

SpikZASA, 2024-09-30:
Thank you its working now :)
Sorry to be a pain SD card isn't working as well, I did select it when using web builder.

SpikZASA, 2024-10-01:
@terjeio any joy?

terjeio, 2024-10-02:
> I dont know if this may be related but the SPI speed is it adjustable?

I was not, I have added a symbol [that allows it to be overridden](https://github.com/grblHAL/ESP32/commit/7b7bd8b68cba4f36ed7f8377a9f947dd7f45c7f1) and set it to 10MHz for the xPro.
The board is badly designed since it has to be reduced?

SpikZASA, 2024-10-02:
Thank you Terjeio i re downloaded from web builder and flashed its the same not mounting SD and not showing in webui.

[image]

Does this mean common clock speed or common pin?

SpikZASA, 2024-10-02:
[image]

its works in fluidnc. at the top of this issue i uploaded my config file from fluidnc i tried now to recompile with the same frequency setting but still refuses to load on hal

terjeio, 2024-10-03:
> Does this mean common clock speed or common pin?

Common CS pin for all Trinamic drivers.

I have reworked the SPI code a bit, with some luck this solves the issue. Web Builder has been updated.
Note that according to Espressif docs the SD card has to be initialized first when the SPI bus is shared with other devices.

SpikZASA, 2024-10-03:
[image]
[image]

YES!!!! Thank you so much its reading it now! will try out some files and report back.

SpikZASA, 2024-10-03:
@terjeio not sure if its related to last fix but spindle is no longer working error 14 on m3.

<Idle|MPos:0.000,0.000,0.000|Bf:100,1023|FS:0,0,0>
M3 S10000
ALARM:14
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:100,1023|FS:0,10000,0|A:S>

terjeio, 2024-10-03:
> the spindle is no longer working error 14 on m3.

Odd, I will retest in the morning - it is bedtime here.

SpikZASA, 2024-10-03:
thank you for your patience

SpikZASA, 2024-10-04:
@terjeio did you get a chance as yet?

terjeio, 2024-10-04:
Tests with both a local build and a Web Builder build went well, this with a dev board and a VFD simulator.
What is your `$I` and `$SPINDLES` output?
And if you have the previous build where the SD card was not working can you test again with that?

SpikZASA, 2024-10-04:
$i
[VER:1.1f.20240928:]
[OPT:VNMZHSL2,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,SED,RTC,WIFI,SD,TMC=7]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:146K]
[DRIVER:ESP32]
[DRIVER VERSION:241003]
[DRIVER OPTIONS:4.3.2]
[BOARD:x-Pro v5]
[AUX IO:2,0,0,0]
[WIFI MAC:64:b7:08:c1:f5:f4]
[IP:192.168.0.85]
[NETCON:Websocket]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.09]
[PLUGIN:Trinamic v0.17]
[PLUGIN:MODBUS v0.16]
[PLUGIN::HUANYANG VFD v0.12]
ok
[PLUGIN:WebUI v0.22]
ok
$spindles
ok
0 - Huanyang v1
ok

SpikZASA, 2024-10-04:
i will try previous build shortly

SpikZASA, 2024-10-04:
dont know if this helps but Alarm 14 is active on startup

SpikZASA, 2024-10-04:
sorry i deleted previous webbuilds before sd change :(

SpikZASA, 2024-10-05:
@terjeio I tried to compile older version but I dont think im getting mymachine.h right or other settings.

terjeio, 2024-10-05:
Are you sure the baud rate matches? grblHAL default is 19200, FluidNC 9600.
Other than that I do not understand that the code changes for SPI/SD card should affect the spindle, and it works with my simulator.
What you may try is to flash a stripped down version without the WebUI and SD Card enabled, does that fail as well?

SpikZASA, 2024-10-05:
baud rate i think thats it the web build that i tried when the spindle was working maybe i had it set at 9600. let me try again and adjust settings

SpikZASA, 2024-10-05:
i am so sorry for wasting your time with the spindle story it was incorrect baud rate set on my part. thank you for your understanding and patience and all the fixes you did!

closed 2024-10-05 by SpikZASA