grblHAL/ESP32 #122: Build errors on the last commit

boboxx, 2024-10-01:
In file included from main/main.c:35:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/main.o] Error 1
In file included from main/driver.c:36:
main/./driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
In file included from main/timers.c:23:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/timers.o] Error 1
In file included from main/ioports.c:22:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
In file included from main/uart_serial.c:45:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
main/driver.c: In function 'I2SStepperWakeUp':
main/driver.c:798:5: error: too few arguments to function 'hal.stepper.enable'
     hal.stepper.enable((axes_signals_t){AXES_BITMASK});
     ^~~
*** [.pio/build/esp32/main/ioports.o] Error 1
In file included from main/ioports_analog.c:22:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/uart_serial.o] Error 1
In file included from main/i2c.h:25,
                 from main/i2c.c:22:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/ioports_analog.o] Error 1
*** [.pio/build/esp32/main/i2c.o] Error 1
*** [.pio/build/esp32/main/driver.o] Error 1
In file included from main/ioexpand.c:26:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/ioexpand.o] Error 1
In file included from main/boards/BlackBoxX32.c:24:
main/driver.h:176:17: error: operator '&&' has no left operand
 #if USE_I2S_OUT && STEP_INJECT_ENABLE
                 ^~
*** [.pio/build/esp32/main/boards/BlackBoxX32.o] Error 1
============================== [FAILED] Took 10.73 seconds ==============================

terjeio, 2024-10-02:
[Fixed now](https://github.com/grblHAL/ESP32/commit/7b7bd8b68cba4f36ed7f8377a9f947dd7f45c7f1).

closed 2024-10-02 by terjeio

boboxx, 2024-10-02:
Thank you! it's working :)