grblHAL/ESP32 #133: Trinamic 2130 drivers on the BDRING 6-pack CNC controller v1p5

UniqueLemming, 2025-01-08:
Hi, I'm trying to compile for my 6-pack v1p5. There are several board files for the 6pack boards, but using the web compiler, I wasn't able to initialize my drivers. The map files that are there don't have separate CS pins, which I believe this board does need. So I've compiled it myself with my own board map. However, I am still not able to initialize the drivers. 
There is a good chance that I'm just doing something wrong, but do you have any idea what?

M122I
[MSG:Warning: Could not communicate with stepper driver!]

When I try to change the current value at $140, I get:
error:2 - Missing the expected G-code word value or numeric value format is not valid.

I'll add: My console output from IO Sender, platformio.ini, my_machine.h, my_machine_map.h

UniqueLemming, 2025-01-08:
/*
  bdring_i2s_6_axis_map.h - An embedded CNC Controller with rs274/ngc (g-code) support

  Driver code for ESP32

  Part of grblHAL

  Copyright (c) 2020-2024 Terje Io

  grblHAL is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  grblHAL is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with grblHAL. If not, see <http://www.gnu.org/licenses/>.
*/

#define BOARD_NAME "Sam's 6Pack CNC V1p5"
//#define BOARD_NAME "BDRING 6-axis I2S"

#if KEYPAD_ENABLE == 1
#error No free pins for I2C keypad!
#endif

#include "use_i2s_out.h"

#if SDCARD_ENABLE || TRINAMIC_SPI_ENABLE

// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
#define PIN_NUM_MISO            19
#define PIN_NUM_MOSI            23
#define PIN_NUM_CLK             18
#if SDCARD_ENABLE
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_CS              5
#endif // SDCARD_ENABLE
#endif // SDCARD_ENABLE || TRINAMIC_SPI_ENABLE

#define I2S_OUT_BCK             GPIO_NUM_22
#define I2S_OUT_WS              GPIO_NUM_17
#define I2S_OUT_DATA            GPIO_NUM_21

#if TRINAMIC_SPI_ENABLE
#define MOTOR_CSX_PIN           I2SO(3)
#define MOTOR_CSY_PIN           I2SO(6)
#define MOTOR_CSZ_PIN           I2SO(11)
#endif

#define X_STEP_PIN              I2SO(2)
#define X_DIRECTION_PIN         I2SO(1)
#define X_ENABLE_PIN            I2SO(0)
#define X_LIMIT_PIN             GPIO_NUM_33 //Module 1-1

#define Y_STEP_PIN              I2SO(5)
#define Y_DIRECTION_PIN         I2SO(4)
#define Y_ENABLE_PIN            I2SO(7)
#define Y_LIMIT_PIN             GPIO_NUM_32 //Module 1-2

#define Z_STEP_PIN              I2SO(10)
#define Z_DIRECTION_PIN         I2SO(9)
#define Z_ENABLE_PIN            I2SO(8)
#define Z_LIMIT_PIN             GPIO_NUM_12 //Module 5-3 - I use probe instead

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS >= 1
#define M3_AVAILABLE
#define M3_STEP_PIN             I2SO(13)
#define M3_DIRECTION_PIN        I2SO(12)
#define M3_ENABLE_PIN           I2SO(15)
#define M3_LIMIT_PIN            GPIO_NUM_35 //Module 1-3
#if TRINAMIC_SPI_ENABLE
#define MOTOR_CSM3_PIN          I2SO(14)
#endif // TRINAMIC_SPI_ENABLE
#endif // N_ABC_MOTORS >= 1

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS >= 2
#define M4_AVAILABLE
#define M4_STEP_PIN             I2SO(18)
#define M4_DIRECTION_PIN        I2SO(17)
#define M4_ENABLE_PIN           I2SO(16)
#define M4_LIMIT_PIN            I2SO(24)    //Module 5-1
#if TRINAMIC_SPI_ENABLE
#define MOTOR_CSM4_PIN          I2SO(19)
#endif  // TRINAMIC_SPI_ENABLE
#endif  // N_ABC_MOTORS >= 2

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 3
#define M5_AVAILABLE
#define M5_STEP_PIN             I2SO(21)
#define M5_DIRECTION_PIN        I2SO(20)
#define M5_ENABLE_PIN           I2SO(23)
#define M5_LIMIT_PIN            I2SO(25)    //Module 5-2
#if TRINAMIC_SPI_ENABLE
#define MOTOR_CSM5_PIN          I2SO(22)
#endif  // TRINAMIC_SPI_ENABLE
#endif  // N_ABC_MOTORS == 3

#define AUXOUTPUT0_PIN          GPIO_NUM_26  // Module 3-1  -  Spindle PWM
#define AUXOUTPUT1_PIN          GPIO_NUM_4   // Module 3-2  -  Spindle enable
#define AUXOUTPUT2_PIN          GPIO_NUM_16  // Module 3-3  -  Coolant mist
#define AUXOUTPUT3_PIN          GPIO_NUM_27  // Module 3-4  -  Coolant flood

// Define driver spindle pins
#if DRIVER_SPINDLE_ENABLE & SPINDLE_PWM
#define SPINDLE_PWM_PIN         AUXOUTPUT0_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_DIR
#define SPINDLE_DIRECTION_PIN   AUXOUTPUT4_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_ENA
#define SPINDLE_ENABLE_PIN      AUXOUTPUT1_PIN
#endif

// Define flood and mist coolant enable output pins.
#define COOLANT_MIST_PIN       AUXOUTPUT2_PIN    // M7  -  Module 3-3
#define COOLANT_FLOOD_PIN      AUXOUTPUT3_PIN    // M8  -  Module 3-4

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
//#define RESET_PIN               GPIO_NUM_34
#define FEED_HOLD_PIN           GPIO_NUM_25    //Module 2-2
#define CYCLE_START_PIN         GPIO_NUM_2     //Module 2-1

#define AUXINPUT0_PIN           GPIO_NUM_39    //Module 2-3
#define AUXINPUT1_PIN           GPIO_NUM_36    //Module 2-4
#define AUXINPUT2_PIN           GPIO_NUM_34    //Module 1-4   -  Probe

#if PROBE_ENABLE
#define PROBE_PIN               AUXINPUT2_PIN
#endif

#ifdef ADD_SERIAL1
#define SERIAL1_PORT
#define UART1_RX_PIN            GPIO_NUM_15
#define UART1_TX_PIN            GPIO_NUM_14
#if MODBUS_ENABLE & MODBUS_RTU_DIR_ENABLED
#define MODBUS_DIRECTION_PIN    GPIO_NUM_13
#endif
#endif

// Module 1-1 - GPIO_NUM_33
// Module 1-2 - GPIO_NUM_32
// Module 1-3 - GPIO_NUM_35
// Module 1-4 - GPIO_NUM_34

// Module 2-1 - GPIO_NUM_2
// Module 2-2 - GPIO_NUM_25
// Module 2-3 - GPIO_NUM_39
// Module 2-4 - GPIO_NUM_36

// Module 3-1 - GPIO_NUM_26
// Module 3-2 - GPIO_NUM_4
// Module 3-3 - GPIO_NUM_16
// Module 3-4 - GPIO_NUM_27

// Module 4-1 - GPIO_NUM_14
// Module 4-2 - GPIO_NUM_13
// Module 4-3 - GPIO_NUM_15
// Module 4-4 - GPIO_NUM_12

// Module 5-1 - I2SO(24)
// Module 5-2 - I2SO(25)
// Module 5-3 - I2SO(26)
// Module 5-4 - I2SO(27)

UniqueLemming, 2025-01-08:
/*
  Part of grblHAL

  Copyright (c) 2020-2024 Terje Io

  grblHAL is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  grblHAL is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with grblHAL. If not, see <http://www.gnu.org/licenses/>.
*/

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used
//#define BOARD_BDRING_V3P5             //
//#define BOARD_BDRING_V4               //
//#define BOARD_BDRING_I2S6A            //
//#define BOARD_BDRING_6X               //
//#define BOARD_BDRING_I2S_6PACK_EXT_V2 //
//#define BOARD_ESPDUINO32              //
//#define BOARD_SOURCERABBIT_4AXIS      //
//#define BOARD_PROTONEER_3XX           //
//#define BOARD_FYSETC_E4               //
//#define BOARD_XPRO_V5                 //
//#define BOARD_MKS_DLC32_V2P0          //
//#define BOARD_MKS_TINYBEE_V1          //
//#define BOARD_CNC3040                 //
//#define BOARD_BLACKBOX_X32            // NOTE: Enable in CMakeLists.txt to set board specific defaults for the core!
//#define BOARD_ROOTCNC_V2              //
//#define BOARD_ROOTCNC_V3              //
//#define BOARD_ROOTCNC_PRO             //
//#define BOARD_JACKPOT                 // Uses TMC2209 drivers, untested!
//#define BOARD_BTT_RODENT              // Uses TMC2160 drivers, untested!
//#define BOARD_CNC_BOOSTERPACK         //
//#define BOARD_GENERIC_S3              // Generic map for ESP32-S3, untested - WIP!
//#define BOARD_GENERIC_I2S_S3          // Generic map for ESP32-S3 with I2S shift registers for I/O expansion, untested - WIP!
#define BOARD_MY_MACHINE              // Add my_machine_map.h in the boards directory before enabling this!

// Configuration
// Uncomment to enable, for some a value > 1 may be assigned, if so the default value is shown.

#if CONFIG_IDF_TARGET_ESP32S3
#define USB_SERIAL_CDC          1 // Serial communication via native USB.
#endif

// Spindle selection:
// Up to four specific spindle drivers can be instantiated at a time
// depending on N_SPINDLE and N_SYS_SPINDLE definitions in grbl/config.h.
// If none are specified the default PWM spindle is instantiated.
// Spindle definitions can be found in grbl/spindle_control.h.
// More here https://github.com/grblHAL/Plugins_spindle
#define SPINDLE0_ENABLE         SPINDLE_PWM0_NODIR
//#define SPINDLE1_ENABLE         SPINDLE_PWM0_NODIR
//#define SPINDLE2_ENABLE         SPINDLE_NONE
//#define SPINDLE3_ENABLE         SPINDLE_NONE
//#define SPINDLE_OFFSET          1 // Uncomment to enable settings for laser spindle XY-offset.
// **********************
//#define MODBUS_ENABLE           1 // Set to 1 for auto direction, 2 for direction signal on auxiliary output pin.
//#define WEBUI_ENABLE            3 // Enable ESP3D-WEBUI plugin along with networking and SD card plugins.
//#define WEBUI_AUTH_ENABLE       1 // Enable ESP3D-WEBUI authentication.
//#define WIFI_ENABLE             1 //
//#define WIFI_SOFTAP             1 // Use Soft AP mode for WiFi.
//#define ETHERNET_ENABLE         1 // Ethernet streaming. Uses networking plugin.
[... 69 lines cut ...]
#endif
// The following symbols have the default values as shown, uncomment and change as needed.
//#define NETWORK_STA_HOSTNAME    "grblHAL"
//#define NETWORK_STA_IPMODE      1 // 0 = static, 1 = DHCP, 2 = AutoIP
//#define NETWORK_STA_IP          "192.168.5.1"
//#define NETWORK_STA_GATEWAY     "192.168.5.1"
//#define NETWORK_STA_MASK        "255.255.255.0"
#if WIFI_SOFTAP
//#define NETWORK_AP_SSID         "grblHAL_AP"
//#define NETWORK_AP_PASSWORD     "grblHALap"
//#define NETWORK_AP_HOSTNAME     "grblHAL_AP"
//#define NETWORK_AP_IPMODE       0              // Do not change!
//#define NETWORK_AP_IP           "192.168.4.1"  // Do not change!
//#define NETWORK_AP_GATEWAY      "192.168.4.1"  // Do not change!
//#define NETWORK_AP_MASK         "255.255.255.0"
#endif
//#define NETWORK_FTP_PORT     21
//#define NETWORK_TELNET_PORT  23
//#define NETWORK_HTTP_PORT    80
#if HTTP_ENABLE
//#define NETWORK_WEBSOCKET_PORT  81
#else
//#define NETWORK_WEBSOCKET_PORT  80
#endif // HTTP_ENABLE
#endif // WIFI_ENABLE

UniqueLemming, 2025-01-08:
#
# Building for BDring new 6x board
#

[platformio]
src_dir = main
include_dir = main

[wifi_networking]
build_flags =

[env:esp32dev]
platform = espressif32 @ ~4.0.0
framework = espidf
board = esp32dev
board_build.partitions = partitions.csv

board_build.embed_files =
    main/embedded/favicon.ico
    main/embedded/ap_login.html
    main/embedded/index.html.gz

build_flags =
    -Wimplicit-fallthrough=1
    -Wno-missing-field-initializers
    -Wno-maybe-uninitialized
    -Wno-stringop-truncation

monitor_speed=115200

lib_compat_mode = off

UniqueLemming, 2025-01-08:
[VER:1.1f.20250107:]
[OPT:VNMDSL2,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,REBOOT,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:235K]
[DRIVER:ESP32]
[DRIVER VERSION:241222]
[DRIVER OPTIONS:4.3.2]
[BOARD:Sam's 6Pack CNC V1p5]
[AUX IO:2,0,0,0]
[PLUGIN:Trinamic v0.21]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
M122I
[MSG:Warning: Could not communicate with stepper driver!]
error:2 - Missing the expected G-code word value or numeric value format is not valid.
$Pins
[PIN:25,Feed hold]
[PIN:2,Cycle start]
[PIN:34,Probe]
[PIN:33,X limit min]
[PIN:35,X limit min 2]
[PIN:32,Y limit min]
[PIN:12,Z limit min]
[PIN:39,Aux in 0,P0]
[PIN:36,Aux in 1,P1]
[PIN:I2S2,X step]
[PIN:I2S13,X2 step]
[PIN:I2S5,Y step]
[PIN:I2S10,Z step]
[PIN:I2S1,X dir]
[PIN:I2S12,X2 dir]
[PIN:I2S4,Y dir]
[PIN:I2S9,Z dir]
[PIN:I2S3,Motor CSX]
[PIN:I2S6,Motor CSY]
[PIN:I2S11,Motor CSZ]
[PIN:I2S14,Motor CSM3]
[PIN:I2S0,X enable]
[PIN:I2S15,X enable]
[PIN:I2S7,Y enable]
[PIN:I2S8,Z enable]
[PIN:4,Spindle on]
[PIN:26,Spindle PWM]
[PIN:16,Mist]
[PIN:27,Flood]
[PIN:19,MISO]
[PIN:23,MOSI]
[PIN:18,SPI CLK]
[PIN:34,RX,Primary UART]
[PIN:35,TX,Primary UART]

terjeio, 2025-01-09:
Individual CS lines for Trinamic drivers was not supported earlier, added now. Note that since the CS lines are routed via I2S a longer delay or some mechanism (I2S passthru mode?) to ensure the lines are actually set properly before transferring data has to be added. I do not have a board available for testing so I am not able to do it.

UniqueLemming, 2025-01-11:
Hi, thanks for the response. I'm still having trouble getting it going. I'm not sure how to how to accomplish the delay you mentioned. It sounds like this particular board might be more trouble than I know how to deal with. Here is what I get now:

M122i
[0;31mE (44890) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44890) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44890) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44900) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44900) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44910) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44920) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44920) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44930) spi_master: check_trans_valid(681): invalid dev handle[0m
[0;31mE (44940) spi_master: check_trans_valid(681): invalid dev handle[0m
[MSG:Warning: Could not communicate with stepper driver!]

Thanks for trying to help.

terjeio, 2025-01-11:
Try this, it works for me when testing the X-axis - in a breadboarded setup with wires all over so less than ideal...

[tmc_spi.zip](https://github.com/user-attachments/files/18386568/tmc_spi.zip)

UniqueLemming, 2025-01-11:
Still getting:
M122I
[MSG:Warning: Could not communicate with stepper driver!]

I'm able to save a new value to $140 now though.

terjeio, 2025-01-12:
Set $338=1 to test with only X enabled, still not working?
And is the motor power on when you try?

Deltaplc, 2025-05-05:
Hi @terjeio
After you helped me make the co2 plugin I started successfully making images and consequently I am pushing myself to the limit with speeds!
So much so that my Arduino Due is taken by the neck, especially using the "jerk"!
I tried to reach 500mm/S but sometimes I have some step losses with my current drivers!

That's why I'm making a shield to insert an esp32 v1 devkit and use the TMC2130 or TMC2160-OC drivers (or something similar)!!
I hope that the esp32 having more "computing power", will better manage the "jerk", and by exploiting all the potential via Spi of the drivers, will be able to reach 500mm/s without losing steps!!
I wanted to use a teensy but they currently cost too much! So I settled on the Esp32, which is cheap and I can easily compile it with Visul Studio Code! My comfort zone....

Since the ESP32 doesn't have many pins, I would have to mount a couple of shift registers (74HC595) on board to be able to drive all the outputs!

Question:
Have the problems of driving the individual drivers (CS) been solved, using the I2S pins as commands?
Can I use TMC2160-OC? Or are they handled differently from the drivers that are currently handled in the dedicated plugin?
Is the syntax for inserting the pin map the one used in the example above by @UniqueLemming ?
In the current pin maps available for esp32 I have not found any examples to follow....I have only found a couple of cases but not with the individual CS.....Are there any pre-established pins?

Thank you in advance for your support!
Greetings

Deltaplc, 2025-05-06:
Thanks for your reply.

> You would have been better off with a RP2350 (Pi Pico2),

Nothing is decided yet....I'm in the process of collecting data and ideas for the project, I will certainly evaluate your advice!
RP2350 (Pi Pico2)
is it already supported by grblhall? Is there anyone who uses it successfully?

> ESP32 family then ESP32 S3 might be a better choice - it has more pins so you should be able to avoid the shift registers.

Maybe I could use the shift registers just to control the drivers, there should be some free pins for Spi data command for the tmc2130..

> Not sure what you mean by this.

I meant these... the first ones are depending on the type of drivers, but is it necessary to enable the rest too?

#define TRINAMIC_ENABLE      2130 
//#define TRINAMIC_ENABLE      2209 
//#define TRINAMIC_ENABLE      5160 

//#define TRINAMIC_I2C            1 // Trinamic I2C - SPI bridge interface.
//#define TRINAMIC_DEV            1 // Development mode, adds a few M-codes to aid debugging. Do not enable in production code.

As always, thank you for your support!