grblHAL/ESP32 #138: Loading firmware via Arduino IDE

Deltaplc, 2025-02-02:
Have a good day
Sorry to bother you...
I would need to load GrblHal into an "ESP32 For Wemos D1 Mini For Arduino UNO R3"( Wemos D1 ESP32) coupled to a "CNC Shield V3".
I know my question might seem a little strange....
I wanted to ask if it is possible to upload your firmware with Arduino IDE?
Is there any way to do this??
Thank you in advance
Greetings

terjeio, 2025-02-02:
> Is there any way to do this??

Possibly, you mean upload only not compile? If upload then it may be better to ask in a Arduino forum.

For compilation you have to add at least a .ino but I do not know if that is going to work without changing other stuff.

Deltaplc, 2025-02-02:
> Per la compilazione devi aggiungere almeno un .ino, ma non so se funzionerà senza cambiare altre cose.

Thanks for your reply!
With Arduino IDE it is also possible to load firmware on Esp32 boards....  For example the "Grbl_Esp32" firmware

> Per la compilazione devi aggiungere almeno un .ino, ma non so se funzionerà senza cambiare altre cose.
Yes exactly, you would need an .ino file....for example the "Grbl_Esp32" firmware, has an .ino file inside.....

This is why I asked you the question.....since I have a bit of confidence with the Arduino IDE it made things easier for me....
It doesn't matter... if it complicates your life and you have to change a lot of things, forget it! 
I'll try to upload it another way....
Thank you very much for your time and your courtesy, you are always ready with your answers!!
Thanks again!