grblHAL/ESP32 #139: flash error 16-bin file overlap

Ams-Laser, 2025-02-07:
i get this error while trying to flash
ay help apreciated

terjeio, 2025-02-07:
Incorrect data in the download tool? Adress is 0x1000 for both the boolader and the firmware?

[image]

Ams-Laser, 2025-02-09:
> Incorrect data in the download tool? Adress is 0x1000 for both the boolader and the firmware?
> 
> [image]

the reference in the download tool shows incorrect values using the ones provided worked, Thanks!!!
now opening other issue for the embroidery plugin, hope you can help there too

closed 2025-02-09 by Ams-Laser