grblHAL/ESP32 #145: The controller may accidentally lose setting data

auxp, 2025-02-22:
There were no abnormal operations.All settings data is lost, just like sending $RST*

terjeio, 2025-02-22:
This can happen if NVS memory (often flash) is bad/starting to wear out. Data is checked against a CRC, if this fails settings (and offsets etc.) are reset,

auxp, 2025-02-22:
> 如果 NVS 内存（通常是闪存）损坏/开始磨损，则可能会发生这种情况。根据 CRC 检查数据，如果失败，则重置设置（和偏移量等），

Can this problem be solved? This is unexpected and may occur every few days, weeks, months, or even longer

terjeio, 2025-02-22:
> Can this problem be solved? 

The best option for NVS memory is to use an EEPROM, or even better FRAM since it has practically infinite number of write cycles.
It _could_ be solved by remapping NVS memory to another area of flash if flash wear out is the issue (or implement wear levelling).

auxp, 2025-02-22:
> > 这个问题能解决吗？
> 
> NVS 内存的最佳选择是使用 EEPROM，甚至更好的 FRAM，因为它几乎具有无限数量的写入周期。如果闪存磨损是问题所在，_可以通过_将 NVS 内存重新映射到闪存的另一个区域来解决（或实施磨损均衡）。

Okay, thank you

closed 2025-03-07 by auxp