grblHAL/ESP32 #147: I downloaded the updated ESP32 firmware today, why did the compilation fail

auxp, 2025-03-07:
../main/./grbl/driver_opts2.h:146:2: error: #error "Too many options that uses a serial port are enabled!"
 #error "Too many options that uses a serial port are enabled!"

The previous configuration has not changed
Did you modify anything

auxp, 2025-03-07:
[image]

[image]

auxp, 2025-03-07:
> 哪个板子？

The circuit board I modified was previously compiled normally

terjeio, 2025-03-07:
Info in this [commit](https://github.com/grblHAL/ESP32/commit/ab34bb1c9adf6a1f6e406c85a92de1c2f361e0a7).

closed 2025-03-07 by auxp