grblHAL/ESP32 #149: MKS DLC32 MAX  can't connect to any sender when flashed with Web Builder firmware

bigshug, 2025-03-10:
Hi there.  I can flash the firmware (successfully? - see screenshots) but wont connect to anything - ioSender, Open Builds control, lasergrbl etc.
I can flash it it using Flash Download Tool connecting using ESP32-S3 or the MKS ESP32 Download Tool.

It does connect to ioSender using the firmware provided by MKS but that is GRBL 1.1 - or their version of it - so the board is ok.

Anyone else using grblHAL on this controller?

[image]
[image]

bigshug, 2025-03-10:
Is this relevant when trying to flash using Openbuilds Control?
[ Firmware Upgrade ]  A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

[image]

terjeio, 2025-03-10:
How do you build? I believe the board does not support native USB so you have to try UART via USB<> UART bridge - `(Virtual) UART` if compiling with the Web Builder:

[image]

There is no schematic available so I simply do not know what is corrcet.

> Is this relevant when trying to flash using Openbuilds Control?
[ Firmware Upgrade ] A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

You cannot flash firmware for the ESP32 - the S3 variant is a different chip.

bigshug, 2025-03-10:
Ah - ok.  I assumed since the MKS supplied firmware flashes successfully over USB that this would too?

terjeio, 2025-03-10:
There are two types of USB - for ESP32 all boards uses a UART <> USB converter chip, ESP32-S3 can either use a "native" USB stack handled by the microprocessor or a UART <> USB converter chip.

bigshug, 2025-03-10:
Setting Flash Download Tool to UART seems to have sorted it although I've yet to test it out on a machine. Thanks again for the help.

closed 2025-03-11 by bigshug