grblHAL/ESP32 #158: Problem with pin port use_i2s_out

OrsoLaser, 2025-04-18:
Hi, sorry for the inconvenience.
I'm testing an espduino-32_wemos_d1_r32_uno
With scield V3 cnc
I redid the pin map, because not all the pins were consistent with my board.....
Everything works correctly!

I wanted to add some Aux outputs and I wanted to use "use_i2s_out" with a shift_Register!
Pin map:

#define I2S_OUT_BCK         GPIO_NUM_15   //Pin 11 Shift_Register sn74hc595n
#define I2S_OUT_WS          GPIO_NUM_32   //Pin 12 Shift_Register sn74hc595n
#define I2S_OUT_DATA        GPIO_NUM_33   //Pin 14 Shift_Register sn74hc595n

#define AUXOUTPUT0_PIN      I2SO(1)
#define AUXOUTPUT1_PIN      I2SO(2)
#define AUXOUTPUT2_PIN      I2SO(3)
With the new configuration the doors with schift_Register work, but when I add this configuration the motors do NOT move!!!
I have seen in other pin maps, where it is used (MKS Tinybee V1.0 for example), that the pins are used for motor control 
#define X_STEP_PIN I2SO(1) 
#define X_DIRECTION_PIN I2SO(2) 
#define X_ENABLE_PIN I2SO(0) 

Is there some basic setting that using "use_i2s_out",
comes into conflict, if they are not used for motor control???

Can I find a solution?

OrsoLaser, 2025-04-18:
I'm doing these tests because I would like to create a shield similar to the v3 cnc shield, but tailored for controlling CO2 lasers... And before creating the PCB, I wanted to test which pins to use, since I will then use it exclusively with your GrblHal!!
Actually I wanted to use an "esp32 devkit v1" or something similar to insert into the new shield...
But at the moment I'm using what I have at home to do the first tests!
Then once finished and tested, I will share the project here on github and make it available to the community!

terjeio, 2025-04-19:
If you want to use I2S for the aux signal and not stepper signals then I am afraid the driver has to be changed.

terjeio, 2025-04-20:
> I wanted to add some Aux outputs and I wanted to use "use_i2s_out" with a shift_Register!

If you only want 8 outputs it may be far simpler to [use SPI](https://hackaday.com/2011/11/05/controlling-shift-registers-via-spi/) and add code similar to [this plugin](https://github.com/grblHAL/Plugins_misc/blob/main/pca9654e.c) - use SPI to shift the data out instead of I2C.

OrsoLaser, 2025-04-23:
Thanks for the help!
But I think I'll just use the available pins... I wanted to make something simple and cheap, but functional, to make it as accessible as possible to everyone!
I'll try to use a 38 pin ESP32 DevKit v1 or something similar, the available pins should be enough to control a CO2 laser and also have some extra outputs...
Since I'm making the PCB myself, I'll try to use as many pins as possible from an existing, already available, mapping to avoid problems!
Thanks again