grblHAL/ESP32 #161: Ethernet on Esp32

jnstrem, 2025-05-01:
Let me preface this by saying the firmware supplied by this project is unbelievable in how good it is. 

I've been using this on an esp32 controller for a few years with the generic board layout. I added more equipment to my shop and it seems like I'm starting to get issues on my controller related to electrical noise. My gcode sender is openbuilds control. 

Every once in a while the controller will just stop and I have to hit the resume button and sometimes it will just disconnect from my machine as well. I confirmed my grounding is done properly, dual shielded cables everywhere, opto isolated I/O etc. The longest wires connected to the controller are about 15" to reduce noise as well. 

I've tried connecting over wifi in station mode and over USB with a 2m cable. There seems to be negligible difference between the two. 

My question is can I reassign the spi pins in the generic esp32 boards map from SD card use to use them for Ethernet to spi connection with a w5500 module like the pico controllers? Is there a reason why this can't be done or to my knowledge hasn't been done before? Or would it be better just to get a better, more stable teensy controller or something like that?

terjeio, 2025-05-01:
Lightly tested, and not updated for latest changes, code is [here](https://github.com/grblHAL/ESP32/blob/master/main/enet.c) - so it _is_ possible. No board map exists for it - pins are [defined here](https://github.com/grblHAL/ESP32/blob/5933cd36390748d3776f13917bb0390320263620/main/enet.c#L82-L86). IIRC the test I did was with a temporary map, a dev board and a W5500 breakout connected with dupont wires - not ideal for high speed SPI.

jnstrem, 2025-05-01:
So pretty much all I have to do is enable the Ethernet in the my_machine file, connect the W5500 to the dev board according to pin out and theoretically it should work?

terjeio, 2025-05-01:
> So pretty much all I have to do is enable the Ethernet in the my_machine file, connect the W5500 to the dev board according to pin out and theoretically it should work?

I just enabled the Ethernet option i _CMakeLists.txt_ and _my_machine.h_ - it does not compile, some errors from changed grblHAL code and some from the ESP IDF framework, IIRC Ethernet has to be enabled in the sdkconfig.

jnstrem, 2025-05-01:
In the CMakeLists.txt file I would assume to enable ethernet I would change line 9 to this?
OPTION(Ethernet "Compile with Ethernet support" ON)

I just enabled ethernet in the CMakeLists.txt file and not the my_machine.h file and it was able to compile. No idea if it works or not will have to test when the hardware shows up. 

I opened up the sdkconfig.H and there didn't seem to be anything related to enabling ethernet. The file also told me it was autogenerated and to not edit it.

terjeio, 2025-05-02:
> In the CMakeLists.txt file I would assume to enable ethernet I would change line 9 to this?

Correct, and you have to enable ethernet in [my_machine.h](https://github.com/grblHAL/ESP32/blob/337734739e73f5714fdec9b1764a982a1ec77069/main/my_machine.h#L73) too. I never bothered to make the setup streamlined since I did not hear back from the user that provided the initial PR for ethernet support.

> No idea if it works or not will have to test when the hardware shows up.

I have just commited changes to make it compile. Don't know if it works though.

> I opened up the sdkconfig.H and there didn't seem to be anything related to enabling ethernet. The file also told me it was autogenerated and to not edit it.

Use [menuconfig](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/kconfig/project-configuration-guide.html) to edit _sdkconfig_. Ethernet options are in _Component config -> Ethernet ->_

I use the tool provided in the Espressif-IDE:

[image]

Anyway, here is the one I used to compile:

[sdkconfig.zip](https://github.com/user-attachments/files/20008718/sdkconfig.zip)

jnstrem, 2025-05-07:
I haven't been able to get the firmware to compile on my end. 
I have been using visual studio with the platform io extension to compile. There are several places where the sdkconfig files exist. I have tried to use the sdkconfig that you gave me to, to overwrite these to no avail.
Am I going to have to switch to the Espressif-IDE: to get this to work? Or am I just not seeing where I am supposed to make this change?

terjeio, 2025-12-23:
Did you manage to get ethernet working?

stefan63, 2026-10-25:
Any progress done to get this working?
I would love to be able to get this board to play with grblHAL over ethernet.

[image]

terjeio, 2026-10-25:
The new [Ooznest controller](https://github.com/grblHAL/ESP32/blob/master/main/boards/ooznest_cnc_map.h) has an ESP32-S3 MCU and Ethernet so it is possible. A board map is likely needed for this board, you can make one? Ethernet has to be enabled in the [sdkconfig](https://github.com/grblHAL/ESP32/blob/0b2d3e1bb7da3b0ae0e03490b71275374eb84c67/sdkconfig.ooznest-cnc#L345-L354) if you want to try.