grblHAL/ESP32 #162: Struggle to compile

smartes, 2025-05-05:
Im fighting with compilation. I want to compile the source for custom ESP32S3 board (using platformio).
 I got the list of errors. How to resolve this issue. Maybe there is updated instruction for newer IDF?
```

main/driver.c:3940:5: error: 'TIMERG0' undeclared (first use in this function); did you mean 'TIMER_0'?

main/uart_serial.c:302:5: error: implicit declaration of function 'periph_module_reset' [-Wimplicit-function-declaration]

main/uart_serial.c:302:26: error: 'periph_module_t' undeclared (first use in this function); did you mean 'periph_port_t'?

main/uart_serial.c:302:43: error: 'PERIPH_UART0_MODULE' undeclared (first use in this function)

main/uart_serial.c:303:5: error: implicit declaration of function 'periph_module_enable' [-Wimplicit-function-declaration]

main/uart_serial.c:56:37: error: 'ETS_UART0_INTR_SOURCE' undeclared (first use in this function); did you mean 'UART_INTR_SOURCE'?

main/uart_serial.c:56:68: error: 'ETS_UART1_INTR_SOURCE' undeclared (first use in this function); did you mean 'UART_INTR_SOURCE'?

main/uart_serial.c:56:98: error: 'ETS_UART2_INTR_SOURCE' undeclared (first use in this function); did you mean 'UART_INTR_SOURCE'?
```

terjeio, 2025-05-06:
> Maybe there is updated instruction for newer IDF?

Newer than 4.4? If so I have not bothered to update since it might just add many hours of work to fix/revalidate the code for apparently no good reason. And I have not checked if there are any instructions provided by Espressif for updating.

dylanetaft, 2025-05-25:
Ran into this too, I am going to download the 4.4 SDK, but it seems like it's well out of the support window.