grblHAL/ESP32 #164: TinyBee + Web Builder -> CPU Halted

simitli, 2025-05-11:
Hello. I'm not able to compile code for MKS TinyBee v1.0 via Web Builder

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12792
load:0x40080400,len:3040
entry 0x400805e4
error:7
main/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008172d  PS      : 0x00060930  A0      : 0x800e28d2  A1      : 0x3ffb6130
A2      : 0x3ffc2c0c  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffc460c
A6      : 0x00000000  A7      : 0x00002000  A8      : 0x00001249  A9      : 0x00000000
A10     : 0xffffffbf  A11     : 0x00000249  A12     : 0x0000004c  A13     : 0x3ffc2c58
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x4008172a:0x3ffb6130 0x400e28cf:0x3ffb6150 0x400d46a7:0x3ffb6170 0x400d1f8b:0x3ffb61b0

ELF file SHA256: 9100c1af871e7a43

CPU halted.

terjeio, 2025-05-11:
Try erasing the flash before flashing the firmware.

simitli, 2025-05-11:
I have done erasing but not help

terjeio, 2025-05-11:
I get the error on the first startup after flashing (with a dev board), after power cycling it is ok.

simitli, 2025-05-11:
After power no "error:7" but still CPU halted

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12792
load:0x40080400,len:3040
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008172d  PS      : 0x00060930  A0      : 0x800e28d2  A1      : 0x3ffb6130  
A2      : 0x3ffc2c0c  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffc460c  
A6      : 0x00000000  A7      : 0x00002000  A8      : 0x00001249  A9      : 0x00000000  
A10     : 0xffffffbf  A11     : 0x00000249  A12     : 0x0000004c  A13     : 0x3ffc2c58  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008172a:0x3ffb6130 0x400e28cf:0x3ffb6150 0x400d46a7:0x3ffb6170 0x400d1f8b:0x3ffb61b0

ELF file SHA256: 9100c1af871e7a43

CPU halted.

terjeio, 2025-05-11:
Save the board in the Web Builder and post the file after zipping it.

[image]

simitli, 2025-05-11:
[ESP32_MKS_TINYBEE_V1.zip](https://github.com/user-attachments/files/20149608/ESP32_MKS_TINYBEE_V1.zip)

terjeio, 2025-05-11:
Thanks, you can try a new build now.

simitli, 2025-05-11:
It's working now, thank you.

closed 2025-05-11 by simitli