grblHAL/ESP32 #170: Firmware for DLC32 Max dit not work

Kristian010, 2025-06-27:
Hi, i have tryed to generate a grblhal Firmware with the Web builder for the DLC32 Max controller. 
I have combined the 3 files for flashing. After flashing the bin file the grblhal won‘t work. 
Have some one a correct bin file for me. I will use it in 50 Watt CO2 laser.
Thanks if someone can help me.

terjeio, 2025-06-27:
Works for me when using the ESP32 download tool:

[image]

[image]

It might be a sender issue, depending on how the UART handshake signals (DTR/RTS) are set on connect the controller may end up in download mode instead of running grblHAL. E.g. with ioSender I have to toggle DTR:

[image]

Kristian010, 2025-06-27:
Hi, many thanks for your fast reply.
I will try it again with your explanation and hope
I get it to work. I see you have use a different usb driver CH343 my one was CH340 is thats maybe the Problem?

osiasfx81-hub, 2025-08-29:
Hi. I have the exact same problem with Kristian010.

I got the MKS DLC 32 MAX board which came with Grbl 1.1h firmware (filesize=8MB) installed. I downloaded the grblHal firmware (filesize=396KB) from web builder, installed it and the board does not respond.

Is there something missing from the grblHal firmware?

terjeio, 2025-08-29:
I just downloaded and flashed to a dev board (with default options) and that boots for me:

[VER:1.1f.2020829:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,PROBES=1,ES,REBOOT,TC,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:EP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:337K]
[DRIVER:ESP32-S3]
[DRIVER VERSION:250706]
[DRIVER OPTIONS:4.4.2]
[BOARD:MKS DLC32 MAX 1.0]

I do not have a MAX board to test with, can you connect with a terminal app (like PuTTY) and briefly short the reset pins (J2). Then post the output you get.

terjeio, 2025-09-10:
No, I've tested with a scope and motors connected to a dev board and not seeing issues.
It might help if you would post your settings (`$$` output) and config (`$I` output) - and also a small test file that do not work.

osiasfx81-hub, 2025-12-06:
ok your explanation is quite logical. do you think there is a way around this or is it a limit of the driver board capacity to process it quickly enough so you can't notice it?

thanks again very much for your knowledgable responses!

osiasfx81-hub, 2026-01-13:
Another question if I may: can I use the bluetooth connection with grblhal with the mks dlc32 max and if so how because when I look at grtbl settings there is no option for bluetooth only wifi. thank you

osiasfx81-hub, 2026-01-13:
> The ESP Bluetooth library is different for the S3 and I have not added a
> driver on top if yet. I do have some code from two years ago that I cannot
> remember the status of. If you compile yourself and want to try it/complete
> it I can post it in a comment.

No unfortunately I am not a programmer😭, I am just a novice trying to use
my free time creatively😂! Anyway thank you so much for your time and your
quick response.
You've done a fantastic job with grblHal web builder.
I can now use my laser engraver for more accurate layouts thanks to your
software and backlash compensation!
Many many thanks!
Take care

osiasfx81-hub, 2026-01-20:
One final thing i noticed is that wifi connection doesn't work with
grblhal where as it works with original grbl. Is there a setting/settings
in grblhal web builder that I should select aside wifi?
My router doesn't seem to recognize mks dlc32 max wifi with grblhal firmware
Thank you

terjeio, 2026-01-22:
What is not working? You cannot connect to the router (no IP)?

> in grblhal web builder that I should select aside wifi?

You should select the protcols to enable. E.g. if you just want to connect via the network enable _Telnet server_.